TerosTechnology/terosHDL #19: Uncaught TypeError: atom.addInfo is not a function

miromax42, 2019-04-02:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.35.1 x64
**Electron**: 2.0.18
**OS**: Unknown Windows version
**Thrown From**: [TerosHDL](https://github.com/TerosTechnology/terosHDL) package 1.0.1

### Stack Trace

Uncaught TypeError: atom.addInfo is not a function

```
At C:\Users\User\.atom\packages\TerosHDL\lib\run\simulator\runSimulator.js:35

TypeError: atom.addInfo is not a function
    at runSimulator (/packages/TerosHDL/lib/run/simulator/runSimulator.js:35:10)
    at runTest (/packages/TerosHDL/lib/run/simulator/main.js:36:3)
    at HTMLElement.TerosHDLRunTest (/packages/TerosHDL/lib/main.js:75:34)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.35.1/resources/app/static/<embedded>:11:350017)
    at CommandRegistry.dispatch (~/AppData/Local/atom/app-1.35.1/resources/app/static/<embedded>:11:348492)
    at ToolBarButtonView.executeCallback (/packages/tool-bar/lib/tool-bar-button-view.js:130:21)
    at ToolBarButtonView._onClick (/packages/tool-bar/lib/tool-bar-button-view.js:109:12)
```

### Commands

```
     -1:56.1.0 TerosHDL:run-test (input.hidden-input)
     -1:17.4.0 intentions:highlight (input.hidden-input)
     -1:17.4.0 settings-view:open (input.hidden-input)
     -0:03.6.0 TerosHDL:run-test (input.hidden-input)
```

### Non-Core Packages

```
aligner-vhdl 0.1.0 
ask-stack 2.2.0 
atom-beautify 0.33.4 
atom-console 0.4.6 
atom-ide-ui 0.13.0 
atom-material-syntax 1.0.8 
atom-material-ui 2.1.3 
atom-military-style-ui 1.2.0 
autocomplete-paths 2.12.2 
busy-signal 2.0.1 
city-lights-syntax 1.1.8 
city-lights-ui 1.5.3 
console-panel 0.3.1 
file-icons 2.1.31 
ghdl-interface 2.0.0 
git-plus 8.6.2 
highlight-selected 0.16.0 
ide-vhdl 0.4.0 
intentions 1.1.5 
language-vhdl 0.7.2 
linter 2.3.0 
linter-quartus 0.2.1 
linter-ui-default 1.7.1 
linter-vhdl 1.1.4 
minimap 4.29.9 
monokai 0.24.0 
monokai-seti 0.9.0 
open-recent 5.0.0 
pane-layout-plus 0.7.4 
seti-syntax 1.2.0 
seti-ui 1.11.0 
seti-vs-syntax 0.1.1 
styri-syntax 1.0.0 
syrinx-dark-syntax 0.4.0 
teletype 0.13.3 
terminal-plus 0.14.5 
TerosHDL 1.0.1 
tool-bar 1.1.11 
vaporwave-ui 0.0.2 
vhdl-entity-converter 0.9.0 
vhdl-signal-definition-finder 1.0.4 
vice-city-syntax 0.2.0 
```

qarlosalberto, 2019-04-02:
We have already fixed the bug.

But I'm sorry, we do not support Windows currently. We will support Windows soon.

It is a started guide: https://www.youtube.com/watch?v=tgr1KGIitIQ

#20

qarlosalberto, 2019-05-19:
@miromax42 the new version supports Windows ;)

closed 2019-05-19 by qarlosalberto