TerosTechnology/terosHDL #24: Error finding libraries in testbench..

jaimenj, 2019-04-27:
Using Ubuntu 18, Ghdl, prueba.vhd is:

`library IEEE;
use ieee.std_logic_1164.all;

entity prueba is
    port (
        F: out std_logic;
        G: out std_logic;
        a: in std_logic;
        b: in std_logic;
        c: in std_logic
    );
end entity prueba;

architecture prueba of prueba is
begin
    F <= (a and b) or not c;
    G <= (a and b) or (not b and c);
end architecture prueba;`

Using TerosHDL, I generate the prueba_tb.vhd file. I get this:

`--! Standard library.
library ieee;
--! Logic elements.
use ieee.std_logic_1164.all;
--! Arithmetic functions.
use ieee.numeric_std.all;
--
library std;
use std.textio.all;
--
library src_lib;
-- use src_lib.types_declaration_prueba_pkg.all;
-- vunit
library vunit_lib;
context vunit_lib.vunit_context;
-- use vunit_lib.array_pkg.all;
-- use vunit_lib.lang.all;
-- use vunit_lib.string_ops.all;
-- use vunit_lib.dictionary.all;
-- use vunit_lib.path.all;
-- use vunit_lib.log_types_pkg.all;
-- use vunit_lib.log_special_types_pkg.all;
-- use vunit_lib.log_pkg.all;
-- use vunit_lib.check_types_pkg.all;
-- use vunit_lib.check_special_types_pkg.all;
-- use vunit_lib.check_pkg.all;
-- use vunit_lib.run_types_pkg.all;
-- use vunit_lib.run_special_types_pkg.all;
-- use vunit_lib.run_base_pkg.all;
-- use vunit_lib.run_pkg.all;

entity prueba_tb is
  --vunit
  generic (runner_cfg : string);
end;

architecture bench of prueba_tb is

  -- Generics
  -- clock period
  constant clk_period : time := 5 ns;
  -- Signal ports
    signal F : std_logic;
  signal G : std_logic;
  signal a : std_logic;
  signal b : std_logic;
  signal c : std_logic;

begin
  -- Instance
  prueba_i : entity src_lib.prueba
  port map (
    F => F,
    G => G,
    a => a,
    b => b,
    c => c
  );

  main : process
  begin
    test_runner_setup(runner, runner_cfg);
    while test_suite loop
      if run("test_alive") then
        info("Hello world test_alive");
        wait for 100 ns;
        test_runner_cleanup(runner);

      elsif run("test_0") then
        info("Hello world test_0");
        wait for 100 ns;
        test_runner_cleanup(runner);
      end if;
    end loop;
  end process;

  clk_process :process
  begin
    clk <= '1';
     wait for clk_period/2;
     clk <= '0';
     wait for clk_period/2;
  end process;

end;`

But finally linter says me this:

`Error	Vhdl Linter	cannot find resource library "src_lib"	11:9
Error	Vhdl Linter	cannot find resource library "vunit_lib"	14:9
Error	Vhdl Linter	context clause not allowed before vhdl 08	15:1
Error	Vhdl Linter	entity 'prueba_tb' was not analysed	37:14`

I cannot find how to solve this. What can I do for installing those missing libraries?¿

qarlosalberto, 2019-04-30:
Hi @jaimenj 

You must have installed: 

TerosHDLbackend: ```pip install TerosHDL```
VUnit: ```pip install vunit_hdl ```

And update terosHDL to the latest version.

To run a testbench:
1. Create a module. (your prueba.vhd).
2. Create a tb. (your prueba_tb.vhd).
3. Create a run.py VUnit file:

[image]

4. Simulation config:

[image]

5. Run test :)

qarlosalberto, 2019-04-30:
Your linter errors:

```
Error Vhdl Linter cannot find resource library "src_lib" 11:9
Error Vhdl Linter cannot find resource library "vunit_lib" 14:9
```
If you install VUnit it insn't a problem.

```
Error Vhdl Linter context clause not allowed before vhdl 08 15:1
```
The file uses VHDL 2008, but not your linter.

You have that errors in your linter, but they will not affect the simulation.

jaimenj, 2019-04-30:
Thank you very much! But avoiding those linter errors, I continue having others :(
When I try to configure the simulation, an error occurs with message: Error: please, select a correct run.py file.

I think my problem is about my system, I've been upgrading the system since Ubuntu Mate 14. I have Python 2.7 installed, may be the problem is arround these. I will try it on another computer and I will return with the results..

qarlosalberto, 2019-04-30:
Mmmm... Have you selected the run.py?

jaimenj, 2019-04-30:
Yes, TerosHDL helped me to create a prueba_run.py file. After this, I go to the 'Simulation config' button, select display mode, select this prueba_run.py file, and when I push the Select button in the 'Test selection' section I get this error.

The generated prueba_run.py is:

`# -*- coding: utf-8 -*-
from os.path import join , dirname, abspath
import subprocess
from vunit.ghdl_interface import GHDLInterface
from vunit.simulator_factory import SIMULATOR_FACTORY
from vunit   import VUnit, VUnitCLI

##############################################################################
##############################################################################
##############################################################################

#pre_check func
def make_pre_check():
  """
  Before test.
  """
#post_check func
def make_post_check():
  """                            
  After test.                    
  """                            
  def post_check(output_path):   
    check = True                 
    return check                 
  return post_check              

##############################################################################
##############################################################################
##############################################################################

#Check GHDL backend.
code_coverage=False
if( GHDLInterface.determine_backend("")=="gcc" or  GHDLInterface.determine_backend("")=="GCC"):
  code_coverage=True
else:
  code_coverage=False

#Check simulator.
print "============================================="
simulator_class = SIMULATOR_FACTORY.select_simulator()
simname = simulator_class.name
print simname
if (simname == "modelsim"):
  f= open("modelsim.do","w+")
  f.write("add wave * \nlog -r /*\nvcd file\nvcd add -r /*\n")
  f.close()
print "============================================="

##############################################################################
##############################################################################
##############################################################################

#VUnit instance.
ui = VUnit.from_argv()

##############################################################################
##############################################################################
##############################################################################

#Add array pkg.
ui.add_array_util()

#Add module sources.
prueba_src_lib = ui.add_library("src_lib")
prueba_src_lib.add_source_files("prueba.vhd")

#Add tb sources.
prueba_tb_lib = ui.add_library("tb_lib")
prueba_tb_lib.add_source_files("prueba_tb.vhd")

#func checks
tb_generated = prueba_tb_lib.entity("prueba_tb")
for test in tb_generated.get_tests():
  test.add_config(name="prueba", pre_config=make_pre_check(),post_check=make_post_check())

##############################################################################
##############################################################################
##############################################################################

#GHDL parameters.
if(code_coverage==True):
  prueba_src_lib.add_compile_option   ("ghdl.flags"     , [  "-fprofile-arcs","-ftest-coverage" ])
  prueba_tb_lib.add_compile_option("ghdl.flags"     , [  "-fprofile-arcs","-ftest-coverage" ])
  ui.set_sim_option("ghdl.elab_flags"      , [ "-Wl,-lgcov" ])
  ui.set_sim_option("modelsim.init_files.after_load" ,["modelsim.do"])
else:
  ui.set_sim_option("modelsim.init_files.after_load" ,["modelsim.do"])

ui.set_sim_option("disable_ieee_warnings", True)

#Run tests.
try:
  ui.main()
except SystemExit as exc:
  all_ok = exc.code == 0

#Code coverage.
if all_ok:
  if(code_coverage==True):
    subprocess.call(["lcov", "--capture", "--directory", "prueba.gcda", "--output-file",  "code_0.info" ])
    subprocess.call(["genhtml","code_0.info","--output-directory", "html"])
  else:
    exit(0)
else:
  exit(1)
`

qarlosalberto, 2019-04-30:
It runs for me. Can you try?:

```
cd /path/to/runpy/
python vunit prueba_run.py 
```
And show me your log.

smgl9, 2019-04-30:
I would ask the same as @qarlosalberto It seems to be an error in the run.py file. 
```
cd /path/to/runpy/
python prueba_run.py 
```
I suggest you try to create the run.py file without any checks unless you plan to edit the the file later and create a complex testbench. In that case you can refer to vunit documentation. This is because we may have not tested yet all the possible check cases.

jaimenj, 2019-05-02:
Bufff :( I think that I must learn a lot about all of those tools before continuing (Ghdl, VUnit, GtkWave..).
But the fact is that TerosHDL is a great project!

I've formatted my PC and now prueba_run.py runs in command line, and I can select tests with TerosHDL into Atom. But now, I think I forget something to install, because I cannot regenerate de prueba_run.py file with other new projects.

It would be nice to have an installation script for Ubuntu, and maybe a sample. I'm really sure that my problem is something missing in the installation process with Ubuntu Mate 18.04.

The output of command 'python prueba_run.py' is:
=============================================
ghdl
=============================================
Re-compile not needed

Starting tb_lib.prueba_tb.prueba.test_alive
Output file: vunit_out/test_output/tb_lib.prueba_tb.prueba.test_alive_8202bbcafb017fc3a78729bbfffd5c6c49d41b75/output.txt
pass (P=1 S=0 F=0 T=2) tb_lib.prueba_tb.prueba.test_alive (0.6 seconds)

Starting tb_lib.prueba_tb.prueba.test_0
Output file: vunit_out/test_output/tb_lib.prueba_tb.prueba.test_0_8aed422a9315a6124f70eb0fb2ce7ef6d1896072/output.txt
pass (P=2 S=0 F=0 T=2) tb_lib.prueba_tb.prueba.test_0 (0.6 seconds)

==== Summary ==============================================
pass tb_lib.prueba_tb.prueba.test_alive (0.6 seconds)
pass tb_lib.prueba_tb.prueba.test_0     (0.6 seconds)
===========================================================
pass 2 of 2
===========================================================
Total time was 1.2 seconds
Elapsed time was 1.2 seconds
===========================================================
All passed!

qarlosalberto, 2019-05-02:
Perfect! I think that you must install the python backend to solve you problem with run.py:

```
pip install TerosHDL
```
Any bug or doubt say it, for us the feedback is very important.

We are working on a next version without dependencies :)

qarlosalberto, 2019-05-02:
Now you can run the tests with the play button ;)

jaimenj, 2019-05-02:
Yesssssssssssssssss!!! Thanks you very much!! I have it completely working.

closed 2019-05-02 by jaimenj