TerosTechnology/terosHDL #29: Please, install correctely TerosHDL backend - Windows

tpougy, 2019-08-21:
Hi, I have installed all requirements for running TerosHDL in windows, but when I click the "Create run.py" button in the toolbar, a message appears with the text:

> Please, install correctely TerosHDL backend
> pip install TerosHDL

The problem is that I already have ran the pip install TerosHDL command in windows power shell, and this text is shown:

> C:\Users\thoma>pip install TerosHDL
> Requirement already satisfied: TerosHDL in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (0.1.1)
> Requirement already satisfied: vunit-hdl>=4.0.0 in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (from TerosHDL) (4.0.8)
> Requirement already satisfied: gitpython>2.1.0 in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (from TerosHDL) (3.0.1)
> Requirement already satisfied: colorama in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (from vunit-hdl>=4.0.0->TerosHDL) (0.4.1)
> Requirement already satisfied: gitdb2>=2.0.0 in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (from gitpython>2.1.0->TerosHDL) (2.0.5)
> Requirement already satisfied: ddt>=1.1.1 in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (from gitpython>2.1.0->TerosHDL) (1.2.1)
> Requirement already satisfied: smmap2>=2.0.0 in c:\users\thoma\appdata\local\programs\python\python37-32\lib\site-packages (from gitdb2>=2.0.0->gitpython>2.1.0->TerosHDL) (2.0.5)
> 

Looks like everything is installed fine, but its not working, how should I proceed?
ps: I supose maybe it is a problem TerosHDL not in windows system variables path

pss: PLEASE HELP ME FAST I NEED TO SEND MY COLLEGE WORK UNTIL SATURDAY!

qarlosalberto, 2019-08-21:
Hi! Can run ```terosHdlRepo --help```  in windows power shell?
What is the output?

qarlosalberto, 2019-08-21:
Do you use python 32 bits?

qarlosalberto, 2019-08-21:
Try to uninstall terohdl python package and install it back again.

tpougy, 2019-08-21:
Hey, I handed my notebook to a friend that is very good with computer bugs and he made it work!
Here is his description of the problem:

"I wanted to know more about the error that was going on. Since I`m not used to debugging Atom add-ons, my first thought was to run it from the command prompt via 'atom --dev --foreground --wait' so I could see its output.

Upon clicking the V large button (which used to spawn the yellow error pop-up), a git error popped up on the console output, informing the user that the call to git's refresh() failed due to a bad or non-existent git executable.

I installed git via the Git for WIndows installer available on the internet with mostly default options (I chose vim instead of nano for the default editor since I know vim's not exactly known for being user friendly, being a vim user myself). It`s very important that you install git allowing it to be on the PATH. That got rid of the errors, apparently.

I recommend you catch the exception when git`s refresh() fails and warn the user the problem's with their git installation, not terosHDL or any of its components."

qarlosalberto, 2019-08-21:
Perfect! Thanks, we will add it as a bug.

qarlosalberto, 2019-08-21:
@tpougy Can you send me an email to carlosruiznaranjo@gmail.com? We are interested in collecting statistics about TerosHDL use to the new features. We would like to ask you about improvements, problems, installation difficulties, documentation... Thank you ;)

closed 2019-09-16 by qarlosalberto