TerosTechnology/terosHDL #31: Installation problem on macOS

woswos, 2019-09-15:
Hello,
I really liked your software and I am trying to install terosHDL to Atom on macOS 10.14.6 and I keep getting the following error message:

> npm ERR! code EBADPLATFORM
npm ERR! notsup Unsupported platform for TerosHDL@1.1.0: wanted {"os":"linux,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm ERR! notsup Valid OS:    linux,win32
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   darwin
npm ERR! notsup Actual Arch: x64
npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/woswos/.atom/.apm/_logs/2019-09-15T19_59_32_442Z-debug.log

Is there a specific reason why macOS is not supported? I can install all of the required dependencies without any problems. 

If further testing on macOS is required, I can help with that as well.

qarlosalberto, 2019-09-16:
Hi!
I think there are no problems to support MacOS, but we currently have no resources to test it.
I have created a branch to test it with MacOS, now you can use it and give us your feedback ;)

https://github.com/TerosTechnology/terosHDL/tree/darwin

woswos, 2019-09-16:
I will test it on macOS but how do I install it manually? I couldn't find any explanation. The procedure might be quite standard and straightforward but I don't know the procedures.

qarlosalberto, 2019-09-16:
You must install Atom and later:

```
git clone https://github.com/TerosTechnology/terosHDL.git -b darwin
cd terosHDL
apm link
apm install .
```

woswos, 2019-09-16:
It installed successfully but the sidebar looks giant:
[image]
I will fork the project and I will try to fix it and other possible problems I see. Let's see how it goes :)

qarlosalberto, 2019-09-16:
@woswos you can resize the menu in the tool-bar package settings:

[image]
We are working on a totally new version. This code will be obsolete promptly :/

smgl9, 2020-04-19:
@woswos 
Meanwhile TerosHDL 2.0 arrives we do some maintenance in TerosHDL 1.  In the release 1.1.7 you should be able to install and use it on macOS.

closed 2020-05-08 by qarlosalberto