TerosTechnology/terosHDL #32: Export EDAM files to use Edalize tools

olofk, 2019-09-16:
[Edalize](https://github.com/olofk/edalize) is a EDA tool abstraction library. If TerosHDL can export EDAM (EDA Metadata) files, it would be able to target any of the (currently 16) backends that Edalize implements, which would quickly bring larger tool support

smgl9, 2019-09-16:
Hi,
In fact TerosHDL is already using Edalize :)
The project has been rebuilt almost from scratch and we expect to publish the code by the end of this week. 

This is the outline of the new version:

[image]

olofk, 2019-09-16:
Haha! Looking forward to hear more about it at ORConf. Guess we can close this then :)