TerosTechnology/terosHDL #33: run script generated "print" without brackets

yamiyukiharu, 2019-09-17:
one of the lines of the generated run.py script has a print statement without brackets (not allowed in python 3). causing the script to fail.
```python
#Check simulator.
print ("=============================================")
simulator_class = SIMULATOR_FACTORY.select_simulator()
simname = simulator_class.name
print simname # <------THIS LINE
if (simname == "modelsim"):
  f= open("modelsim.do","w+")
  f.write("add wave * \nlog -r /*\nvcd file\nvcd add -r /*\n")
  f.close()
print ("=============================================")
```
It worked as soon as I added brackets to them.

smgl9, 2019-09-17:
You are right! 
I'll fix it asap

smgl9, 2019-09-26:
The change is already in the develop branch in the backend repository.

closed 2019-09-26 by smgl9