TerosTechnology/terosHDL #34: "PermissionError: [WinError 5] Access is denied" when trying to run the test

yamiyukiharu, 2019-09-17:
I get the following error while trying to run the generated run script.
Tried running atom as administrator but still no luck.

OS: Windows 10 Home

>=============================================
ghdl
=============================================
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\vunit\ui.py", line 794, in main
    all_ok = self._main(post_run)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\vunit\ui.py", line 838, in _main
    all_ok = self._main_run(post_run)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\vunit\ui.py", line 863, in _main_run
    simulator_if = self._create_simulator_if()
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\vunit\ui.py", line 857, in _create_simulator_if
    output_path=self._simulator_output_path)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\vunit\ghdl_interface.py", line 68, in from_args
    backend=cls.determine_backend(prefix))
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\vunit\ghdl_interface.py", line 103, in determine_backend
    output = subprocess.check_output([join(prefix, "ghdl"), "--version"]).decode()
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python37-32\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied

qarlosalberto, 2019-09-17:
I think it is a error in the installation of vunit. Can you create a run.py file and later?:

```
python run.py 
```

What is the output?

yamiyukiharu, 2019-09-18:
I have no issues running the script if it was done in a separate terminal, but I will get this error whenever I run press the run button in atom

qarlosalberto, 2019-09-18:
I can't reproduce the error. 
1. Can you check the permissions of the folder? C:\Users\admin\AppData\Local\Programs\Python\Python37-32\l
2. Can you install teroshdl backend and vunit with other python version?

yamiyukiharu, 2019-09-19:
I realized that running the test without GTKwave/ModelSim option works fine. I tried changing the folder permissions by unticking read-only option through the properties window, but somehow the read-only option reappears when I inspect the folder properties again.

qarlosalberto, 2019-09-20:
What simulator are you using? Ghdl or Modelsim?

yamiyukiharu, 2019-10-09:
Sorry for the late reply, I am using ghdl for the simulation.