TerosTechnology/terosHDL #36: Uncaught TypeError: Cannot read property 'value' of null

rftafas, 2019-10-01:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.40.1 x64
**Electron**: 3.1.10
**OS**: Ubuntu 16.04.6
**Thrown From**: [TerosHDL](https://github.com/TerosTechnology/terosHDL) package 1.1.0

### Stack Trace

Uncaught TypeError: Cannot read property 'value' of null

```
At /home/ricardo-junior/.atom/packages/TerosHDL/lib/run/simulator/simulator-view.js:331

TypeError: Cannot read property 'value' of null
    at getXml (/packages/TerosHDL/lib/run/simulator/simulator-view.js:331:59)
    at XMLHttpRequest.xmlhttp.onreadystatechange (/packages/TerosHDL/lib/run/simulator/simulator-view.js:293:17)
```

### Commands

```
```

### Non-Core Packages

```
aligner-vhdl 0.1.0 
atom-beautify 0.33.4 
console-panel 0.3.1 
file-icons 2.1.35 
Hydrogen 2.8.0 
language-vhdl 0.7.2 
Sublime-Style-Column-Selection 1.7.5 
tabs-to-spaces 1.0.5 
TerosHDL 1.1.0 
tool-bar 1.1.12 
trailing-spaces 0.4.0 
vhdl-entity-converter 0.9.0 
xml-formatter 0.13.0 
```

smgl9, 2019-10-01:
Hi,
I assume that you are trying to run a Vunit test and it woks if you run it in a console. Could it be possible that you have missed to fill in something in the config dialog?

rftafas, 2019-10-02:
It seems to have been solved by itself after a double check. I'm closing it.

closed 2019-10-02 by rftafas