TerosTechnology/terosHDL #44: therosHDL extension availability for Visual Studio Code

pidgeon777, 2020-02-14:
Hello, I noticed that a terosHDL extension for Atom exists.

Is the same extension also available for Visual Studio Code, or are you planning to release it in the future?

qarlosalberto, 2020-02-14:
We would like to have a version for Visual Studio Code. But it will be in a future, we are only 3 developers and we don't have much time. Now we are working in the version 2.0 that will have a lot of new features.

pidgeon777, 2020-02-14:
Excellent, I'll keep an eye on your future releases.

Providing an extension for Visual Studio Code would be great because the next generation of IDEs will also be compatible with Visual Studio Code extensions.

One example that comes to mind is Eclipse Theia:

https://theia-ide.org/
https://projects.eclipse.org/proposals/eclipse-theia
https://github.com/eclipse-theia/theia

This is a completely customizable editor, in every aspect.

If I wanted to forecast, I'd say that one day this will replace Visual Studio Code (Theia's core is based on this editor).

smgl9, 2020-02-15:
Thank you for your interest. We are aware of Theia.
Now, the project is chained to Atom, but it will be different in version 2.0.
TerosHDL could be available for Atom, VS and other editors. Starting with Atom

pidgeon777, 2020-02-16:
This is a great news. Again, congratulations on your project, keep up the good work!

qarlosalberto, 2020-06-19:
@pidgeon777 We have a beta version of TerosHDL VSCode. If you would you like to be a beta tester send an email to terostechnology@gmail.com