TerosTechnology/terosHDL #46: How do I set modelsim-in?

lukipedio, 2020-03-06:
[image]

qarlosalberto, 2020-03-06:
Go to package configuration: Edit -> Preferences -> Packages -> TerosHDL -> Settings

What TerosHDL version are you using? What OS?

lukipedio, 2020-03-06:
I am using Win 10 Pro 1909 with TerosHDL 1.1.4.

It seems I cannot get an entry of TerosHDL under "packages".
Possible?

Luca

-- 

  CAEN S.p.A.  Luca Colombini

*Frontend Division - Senior Design Engineer*
l.colombini@caen.it
		Tel.: +39 0584 388 398
Fax: +39 0584 388 959

*CAEN S.p.A.*
via Vetraia,11 • 55049 • Viareggio (LU) • Italy
Cod.Fisc. e p.IVA IT00864500467
www.caen.it

AVVISO DI RISERVATEZZA. Le informazioni contenute in questo messaggio di 
posta elettronica e/o files allegati, sono da considerarsi strettamente 
riservati. Il loro utilizzo è consentito esclusivamente al destinatario 
del messaggio, per le finalità indicate nello stesso. Costituisce 
violazione ai principi dettati dalla normativa nazionale ed europea in 
materia di Trattamento dei Dati Personali: trattenere il messaggio 
stesso oltre il tempo necessario, divulgarlo anche in parte, 
distribuirlo ad altri soggetti, copiarlo o utilizzarlo per finalità 
diverse. Ricordiamo che in ogni momento potrete richiederci 
l'informativa per la privacy e la sospensione da parte nostra 
dell'impiego dei vostri dati, ad esclusione delle comunicazioni 
effettuate in esecuzione di obblighi di legge. Qualora riceveste questo 
messaggio senza esserne il destinatario Vi preghiamo cortesemente di 
darcene notizia via e-mail e di procedere alla distruzione del messaggio 
stesso, cancellandolo dal Vostro sistema. Grazie.

The information contained in this e-mail and/or attached files are to be 
considered strictly confidential. Their use is allowed exclusively to 
the recipient of the message, for the indicated purposes. It is a 
violation of the principles dictated by national and European 
legislation on the processing of personal data: retain the message after 
the necessary time, disclose any of its part, distribute it to other 
subjects, copy it or use it for different purposes. We remind you that 
at any time you can request privacy information and the suspension from 
your data use, excluding communications made in compliance with legal 
obligations. If you received this message without being the recipient, 
we kindly ask you to inform us via e-mail and delete it from your 
system. Thank you.

qarlosalberto, 2020-03-06:
I think that it's not possibe.

[image]

lukipedio, 2020-03-06:
Ok, I got it! 
Sorry for this stupid question.
Thanks.

qarlosalberto, 2020-03-06:
There is a little bug, it should load the windows config.

qarlosalberto, 2020-03-06:
https://github.com/TerosTechnology/terosHDL/blob/develop/lib/main.js#L67

lukipedio, 2020-03-06:
There should be a Win config as well. The default path in settings are for Linux

qarlosalberto, 2020-03-06:
Yes, there is a default Windows config https://github.com/TerosTechnology/terosHDL/blob/develop/lib/config/config-schema-windows.json

But the variable has a wrong name... This is de bug :/