TerosTechnology/terosHDL #48: Please, install correctly TerosHDLbackend.

Raulgarza98, 2020-03-30:
Hello, 

I keep getting this error message even though i have already installed the dependencies, what can I do to fix this? I need it for a school assignment due this week.

Here is what I get on Anaconda prompt:

```
(base) C:\Users\raul9>pip install terosHDL
Requirement already satisfied: terosHDL in c:\users\raul9\anaconda3\lib\site-packages (0.1.2)
Requirement already satisfied: vunit-hdl>=4.3.0 in c:\users\raul9\anaconda3\lib\site-packages (from terosHDL) (4.4.0)
Requirement already satisfied: gitpython>2.1.0 in c:\users\raul9\anaconda3\lib\site-packages (from terosHDL) (3.1.0)
Requirement already satisfied: colorama in c:\users\raul9\anaconda3\lib\site-packages (from vunit-hdl>=4.3.0->terosHDL) (0.4.3)
Requirement already satisfied: gitdb<5,>=4.0.1 in c:\users\raul9\anaconda3\lib\site-packages (from gitpython>2.1.0->terosHDL) (4.0.2)
Requirement already satisfied: smmap<4,>=3.0.1 in c:\users\raul9\anaconda3\lib\site-packages (from gitdb<5,>=4.0.1->gitpython>2.1.0->terosHDL) (3.0.1)

(base) C:\Users\raul9>pip install vunit_hdl
Requirement already satisfied: vunit_hdl in c:\users\raul9\anaconda3\lib\site-packages (4.4.0)
Requirement already satisfied: colorama in c:\users\raul9\anaconda3\lib\site-packages (from vunit_hdl) (0.4.3)
```
thanks in advance.

qarlosalberto, 2020-03-30:
What is the error message? Could you send a email to terostechnology@gmail.com?

edsonms, 2020-04-18:
For me (WINDOWS) it solved installing GIT for Windows

qarlosalberto, 2020-05-08:
Can we close the issue?

qarlosalberto, 2021-03-09:
@Zalfrin I recommend using the VSCode version. We have stopped the development for Atom.

https://marketplace.visualstudio.com/items?itemName=teros-technology.teroshdl