TerosTechnology/terosHDL #49: ImportError: No module named vunit.sim_if.ghdl

PronElle, 2020-04-04:
Can't run simulation : when I try to set simulation config> test selection I get the following error from the GUI
```
Error: please select a correct run.py file
```
I did create a run.py file as shown in the tutorial before setting simulation configs, but this is what the console displays
```
ImportError: No module named vunit.sim_if.ghdl
```

qarlosalberto, 2020-04-04:
VUnit has changed the interface. Could you install the last version?

PronElle, 2020-04-04:
As mentioned in the readme I installed vunit_hdl running
```pip3 install vunit_hdl``` 
Here's the output of ```pip3 list | grep vunit``` :
```
vunit (0.1.3)
vunit-hdl (4.4.0)
```

qarlosalberto, 2020-04-04:
What is your TerosHDL python version?
```
pip3 list | grep TerosHDL
```

qarlosalberto, 2020-04-04:
Mm... I think that I know what is the error. Could you install VUnit in Python2 (I know that it sucks, we are working in the new version) ?

PronElle, 2020-04-04:
Here's the output : 
```
TerosHDL (0.1.2)

```
As for python2 , I can try

PronElle, 2020-04-04:
Switching to python2 solves the problem but I can't still run the simulation
```
WARNING - Entity hexa_display_tb matches testbench name regex ^(tb_.*)|(.*_tb)$ but has no generic runner_cfg and will therefore not be run.
in file /home/elle/Desktop/Part4/hex_display_tb.vhd
Traceback (most recent call last):
  File "hex_display_run.py", line 75, in <module>
    tb_generated = hex_display_tb_lib.entity("hex_display_tb")
  File "/home/elle/.local/lib/python2.7/site-packages/vunit/ui/library.py", line 305, in entity
    raise KeyError(name)
KeyError: 'hex_display_tb'
=============================================
ghdl
=============================================
```

qarlosalberto, 2020-04-04:
If you want to use VUnit you must create a compatible testbench. There are 2 ways:
- Create the tb from TerosHDL button.
- Manually. You have an example https://github.com/VUnit/vunit/blob/master/examples/vhdl/run/tb_with_test_cases.vhd

PronElle, 2020-04-04:
K, now I can run my simulation using Modelsim. Nevertheless, when I try to use ghdl instead, I get another error :
```
=============================================
ghdl
=============================================
Compiling into src_lib:   hex_display.vhd                                                                                                
failed
=== Command used: ===
/usr/bin/ghdl -a --workdir=/home/elle/Desktop/Part4/vunit_out/ghdl/libraries/src_lib --work=src_lib --std=08 -P/home/elle/Desktop/Part4/vunit_out/ghdl/libraries/vunit_lib -P/home/elle/Desktop/Part4/vunit_out/ghdl/libraries/src_lib -P/home/elle/Desktop/Part4/vunit_out/ghdl/libraries/tb_lib /home/elle/Desktop/Part4/hex_display.vhd

=== Command output: ===
/usr/bin/ghdl-mcode:warning: library ieee does not exists for v08
/home/elle/Desktop/Part4/hex_display.vhd:1:9: cannot find resource library "ieee"
/home/elle/Desktop/Part4/hex_display.vhd:2:10: unit "std_logic_1164" not found in library "ieee"
/home/elle/Desktop/Part4/hex_display.vhd:3:10: unit "numeric_std" not found in library "ieee"
/home/elle/Desktop/Part4/hex_display.vhd:7:15: no declaration for "signed"
/home/elle/Desktop/Part4/hex_display.vhd:8:18: no declaration for "std_logic_vector"
Compile failed
```
Maybe ghdl is not configured properly. Can you help me with this?

qarlosalberto, 2020-04-04:
Have you the folder **/usr/local/lib/ghdl/ieee/v08** ? Is it empty?

qarlosalberto, 2020-04-04:
Try with the clean button.

PronElle, 2020-04-05:
Actually I don't have the directory 
/usr/local/lib/ghdl/ieee/v08. 
ieee is a different place. 
As for ModelSim, yep the same code runs in modelsim

qarlosalberto, 2020-04-05:
What is your ghdl version and your OS?
ghdl --version

Have you the folder ieee?

qarlosalberto, 2020-04-05:
Are you using ghdl in Debian? https://manpages.debian.org/testing/ghdl-mcode/ghdl-mcode.1.en.html

I think that (for license reasons) it doesn't include the ieee libraries for VHDL 08. You have these options:
1. Compile and install GHDL manually. It's easy for mcode.
```
./configure --prefix=/usr/local
make
sudo make install
```

2. Include the libraries in your project manually (in your run.py): https://github.com/ghdl/ghdl/tree/master/libraries/ieee2008
3. Compile the libraries manually and copy the folder to de ieee folder (I think this also works).

PronElle, 2020-04-23:
I forgot to reply : your last comment solved my problem! Thanks a lot!

closed 2020-04-23 by qarlosalberto