TerosTechnology/terosHDL #51: [Windows] Please, configure correctly your ghdl path

edsonms, 2020-04-18:
Hi,

When trying to Run my test I get a yellow notification message saying:
[image]

How do I config my GHDL path? I already have added it to environment variables.

Kind Regards

edsonms, 2020-04-18:
I have set it here:
[image]
But I keep getting the same message

qarlosalberto, 2020-04-19:
Maybe it's a problem with the white spaces in the path. Try to change ' '
by '\ '

smgl9, 2020-04-19:
Hi @edsonms  

You are not the first reporting exactly the same issue. We will take a look as soon as possible and  let you know.

edsonms, 2020-04-19:
After update to version 1.1.7 the issue was solved.

smgl9, 2020-04-19:
I'm glad to hear that :)

edsonms, 2020-04-19:
Although now I got stuck in the following:
[image]

edsonms, 2020-04-19:
I also get the following error message when I try to select a Test:
[image]

smgl9, 2020-04-19:
ok, 
Mind that you have a button on the toolbar to switch btween simulators, push the button as many times as needed to ensure that you have selected ghdl.

[image]

edsonms, 2020-04-19:
> ok,
> Mind that you have a button on the toolbar to switch btween simulators, push the button as many times as needed to ensure that you have selected ghdl.
> 
> [image]

Hi, it is already selected the GHDL simulator. And the path problem does not happen anymore

edsonms, 2020-04-19:
Solved the problem. Apparently, it was some problem with the path I was using (virtual machine). 

In summary, updating to version 1.1.7 solved the issue original problem

closed 2020-04-19 by edsonms