TerosTechnology/terosHDL #56: Error when creating repository

lawrancechiang, 2020-04-30:
I just installed Atom in which I installed the package TerosHDL.

I tried creating a repository, but it doesn't let me.  I get the error message as follows:

"Please, install correctly TerosHDLbackend.
pip install TerosHDL"

The screenshot is attached.  Please could you assist me with this matter?  Many thanks.

LC 

[image]

qarlosalberto, 2020-04-30:
Have you installed the Python dependency?

pip install TerosHDL

What is your OS?

lawrancechiang, 2020-05-02:
Hi,

Yes, Python 3.8.2 was installed.  I then ran the command "_pip install _TerosHDL_" (cmd.exe of Windows) as shown in the screenshot.  Unfortunately afterwards when I re-attempt to create a project in Atom I see the same message, still asking me to install the TerosHDL backend.

Many thanks.

[image]

qarlosalberto, 2020-05-02:
Have you installed git? Have you git in the system path?

lawrancechiang, 2020-05-02:
I typed "git" into cmd.exe and it seems to me like that is it installed - please see the screenshot below.

[image]

qarlosalberto, 2020-05-02:
Mm.. what is the output if you type: terosHdlRepo --help?

lawrancechiang, 2020-05-02:
It appears that the command doesn't exist!

[image]

smgl9, 2020-05-02:
> Hi,
>
> Yes, Python 3.8.2 was installed. I then ran the command "_pip install _TerosHDL_" (cmd.exe of Windows) as shown in the screenshot. Unfortunately afterwards when I re-attempt to create a project [...]

Hi!,

I'm not sure, but the screenshot of the backend installation may be suggesting us a solution. Take a look at the warning and tell us if adding the directory to the path helps :)

smgl9, 2020-05-07:
@lawrancechiang 

Have you had the opportunity to try the workaround?

lawrancechiang, 2020-05-09:
> 
> 
> @lawrancechiang
> 
> Have you had the opportunity to try the workaround?

Hi smgl9,

Apologies for the delay, and thank you for the suggestion.  I'm not too familiar with cmd.exe - how do I add a directory to the path?

Many thanks,

LC

smgl9, 2020-05-14:
Hi @lawrancechiang 

You can try this: https://stackoverflow.com/questions/9546324/adding-directory-to-path-environment-variable-in-windows