TerosTechnology/terosHDL #67: VUnit lib string_ptr_pkg-body-2002p compilation error

edsonms, 2020-05-17:
Hi,

Working with TerosHDL v1.1.7 on macOS Catalina 10.15.4 and whenever I try to use VUnit test it returns a library compilation error:

> Compiling into vunit_lib: ../../../../../../Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd               
> 
> failed
> === Command used: ===
> /usr/local/bin/ghdl -a --workdir=/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/vunit_lib -P/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/src_lib -P/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/tb_lib /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd

Any ideas guys?

qarlosalberto, 2020-05-17:
Could you re-run with --verbose argument and put the complete output?

edsonms, 2020-05-17:
> Could you re-run with --verbose argument and put the complete output?

Do you mean to add --verbose here?
[image]
It returns the same error as before:

> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/builtins.py:136: Warning: 'array_t' is deprecated and it will removed in future releases;use 'integer_array_t' instead
>   warn(arr_deprecation_note, Warning)
> 
> =============================================
> ghdl
> =============================================
> Compiling into vunit_lib: ../../../../../../Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd         
> 
> failed
> === Command used: ===
> /usr/local/bin/ghdl -a --workdir=/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/vunit_lib -P/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/src_lib -P/Users/edson/Documents/Documentos_Eletronica/Pos_IFSC/Projeto_TC/Radar_idea/FPGA_Flow_Detector/vunit_out/ghdl/libraries/tb_lib /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd
> 
> === Command output: ===
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'm' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'o' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'd' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'e' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character ' ' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'i' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'n' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 't' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'e' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'r' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'n' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'a' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'l' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character ':' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character ' ' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'i' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'd' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character '/' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character '=' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character '-' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character '1' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character ' ' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'n' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:132:32: character 'o' of type character is not visible
[... cut ...]
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:287:31: character 'i' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:287:31: character 'n' of type character is not visible
> /Users/edson/Library/Python/2.7/lib/python/site-packages/vunit/vhdl/data_types/src/string_ptr_pkg-body-2002p.vhd:287:31: character 'g' of type character is not visible
> /usr/local/bin/ghdl: compilation error
> 
> Compile failed