TerosTechnology/terosHDL #71: Dependency issue

logari84, 2020-11-08:
Hello,

I have encountered an error similar to "https://github.com/TerosTechnology/terosHDL/issues/48". In my case I am running Atom on Linux and the problem is present not only for TerosHDLbackend. It seems that I get error for all the dependencies:
- Initialize project/Add module : Please, install correctly TerosHDLbackend (though I installed TerosHDL via pip)
- Create VUnit file : Please, install correctly TerosHDLbackend
- Copy as diagram : Please, install correctly Symbolator. Be careful with the dependencies. (though symbolator is installed)

Could it be an Atom related issue?
Any hint would be highly appreciated!

Thank you in advance.
Stam

smgl9, 2020-11-08:
Hello,

We will check if there is a problem with the latest versions of Atom and try to figure out what's happening. 

Nonetheless Atom version is not including the new developed features and uses the old backend (deprecated). We strongly recommend to use vscode-TerosHDL to keep up with the development of the tool. https://marketplace.visualstudio.com/items?itemName=teros-technology.teroshdl&ssr=false#overview 
Unless you need simulators integration, which it's still not included in vscode version (hopefully soon). 

Eventually Atom TerosHDL will be updated to the new backend.

logari84, 2020-11-09:
Thank you for the feedback.

I have tried it in both Atom and vscode. To be honest I have never worked with these IDEs and I tried them only for TerosHDL. In vscode though I don't see the same options as in Atom (I guess it's still under development and things are being added).

In addition, I get some errors along the code which I can't figure out how to solve. E.g. I loaded the src folder of a VIvado project just to try the IDE and I noticed the error "unit "uart" not found in library "uart"TerosHDL:ghdl(error)". How can I include the custom "uart" library so it is recognized by ghdl?

Thank you.

qarlosalberto, 2020-11-10:
Thank you for the feedback @logari84 

It's a linter limitation. The linter only "know" the current file, it can't relate it with other files. In the next major release we will add a project manager. It will solve the problem.

For now we could add the option to omit the libraries errors for GHDL.

logari84, 2020-11-10:
Good to know. The project manager will be a great addition. Will try to follow the progress so I can help with any useful feedback!

Keep up the good work guys!

closed 2020-11-11 by qarlosalberto