TerosTechnology/terosHDL #84: Uncaught ReferenceError: out_directory is not defined

Lucifer-02, 2021-06-17:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.57.0 x64
**Electron**: 9.4.4
**OS**: Linux Mint
**Thrown From**: [TerosHDL](https://github.com/TerosTechnology/terosHDL) package 1.1.9

### Stack Trace

Uncaught ReferenceError: out_directory is not defined

```
At /home/lucifer/.atom/packages/TerosHDL/lib/run/create/run-py.js:378

ReferenceError: out_directory is not defined
    at CreateRun.createRun (/packages/TerosHDL/lib/run/create/run-py.js:378:20)
    at HTMLButtonElement.CreateRun.doInitButton.onclick (/packages/TerosHDL/lib/run/create/run-py.js:145:12)
```

### Commands

```
  4x -3:06.9.0 core:backspace (input.hidden-input)
     -1:07.2.0 application:add-project-folder (ol.tree-view-root.full-menu.list-tree.has-collapsable-children.focusable-panel)
     -0:58.8.0 TerosHDL:create-run (div.tool-panel.tree-view)
```

### Non-Core Packages

```
console-panel 0.3.1 
language-vhdl 0.7.2 
TerosHDL 1.1.9 
tool-bar 1.4.4 
```

qarlosalberto, 2021-06-17:
I recommend using the VSCode version. We have stopped the development for Atom. You will find a lot of new features :)

https://marketplace.visualstudio.com/items?itemName=teros-technology.teroshdl
https://terostechnology.github.io/terosHDLdoc/

Lucifer-02, 2021-06-18:
thanks a lot!