TerosTechnology/terosHDL #85: where to add library?

marcob19, 2021-06-28:
Hi, 
i use teroshdl for my .hdl codes. As vhdl linter i use vivado. It works but when I map a component using:
<mapp_name> : entity mylib.<entity_name>(<architecture_name>) 
it result as error:
[VRFC 10-149] 'lcd_sender' is not compiled in library customlibTerosHDL:xvhdl(VRFC 10-149)
how to add library location? i can't find any documentation. 
On some vhdl extension it is necessary to generate a file at top of workspace with the library files location, it is the same?

qarlosalberto, 2021-06-28:
The linter only lints the current file. You can try the beta version, it uses a standalone linter, you don't need to configure an external tool as Vivado. And you can use the project manager to create libraries.

https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

qarlosalberto, 2021-06-28:
I recommend using the VSCode version. We have stopped the development for Atom. You will find a lot of new features :)

https://marketplace.visualstudio.com/items?itemName=teros-technology.teroshdl
https://terostechnology.github.io/terosHDLdoc/